BatchDrake/SigDigger #261: How to decode fixed code remote 433MHz?

FrankBolleri, 2024-11-15:
Dears, I have tried lot of SDR software before, but yours seems to me one of the most attracting - it is full of functions and is very fascinating...

I am pretty sure I can use this software to decode a signal from a simple chinese remote modulating in AM/OOK/Manchester code, obtaining the binary and decimal representation of the data, like the 24 bit signal in the figure below:

[image]

But... how? Could you suggest me a way?

Thanks
Francesco

BatchDrake, 2024-11-18:
Hi Frank, and sorry for the late reply. Crazy weekend.

Thanks for your kind words! Yeah, you definitely can, and in many different ways. For short bursts like this one, something I do a lot is the following:

1. In the right panel, one of the options above is the "periodic selection". This divides the selection in several uniformly-spaced intervals you can adjust to the bits (symbols) of the burst. I usually eyeball the number of bits in the burst, fit the selection so that all bits inside are aligned to the subintervals and then increse / decrease the number of intervals until they match the number of bits.
2. In Sampling and decision (for OOK / ASK only), set the decision space to Amplitude, set the sampling interval to Selection, mark "Fixed interval partition" (this is usually autoconfigured to the number of subintervals of the periodic selection) and click on Sample.
3. This will open the sampling window. Above in the window you will see a histogram of the different sampled amplitudes which, if the SNR is good, will look like two bulges (one for the zeroes and the other for the ones). Drag the mouse from the peak of the first bulge to the second bulge and that will adjust the decision threshold.

Other way of doing it is the following, although this one is more prone to errors if the signal is too noisy:

1. Select the burst with the mouse (you don't need to be extremely accurate with this selection, just make sure the selection covers the whole burst as you did in this screenshot)
2. In Sampling and Decision, Amplitude space, Selection, and mark Zero-crossings. Component: absolute value. The pink rectangle drawn behind the waveform marks the decision threshold: anything inside the rectangle is zero, and anything above it is one. Adjust the amplitude in the "Zero point" spinbox until you believe you are in a reasonable threshold.
3. Click sample, etc, etc.

Note that in either case you are going to end with a list of zeros and ones represented as black and white pixels. Those are discretized amplitudes. You can save them to a file as a .txt file or a C array, whatever is more practical to you.

Cheers

FrankBolleri, 2024-11-18:
Thanks for your answer!
I have tested both and first works fine, just seems to me a little tricky.
I will try to explain where the difficults are is in both, if you like to have a feedback.

**METHOD 1**
[image]

I have correctly obtained the value 010101011010101000110110 --> 5614134, as expected.
I really like the "red grid selection method"... is fascinating!
But:
1. I obtain success with "Sync to selection intervals"; to have same success with "Fixed interval partition" I have to take care to put manually 24 in the "Number of symbols".
2. Not clear to me why in "Sampling window" I have to adjust selecting from the first bar to the last - In particular is not clear why I have a first "section", silence, and a second section so distant - and more mistery is why I have to select from start to end. Probably I don't understand the meaning of that graphical tool..
3. I don't understand well the choice to display the grid "chessboard" BLACK for 0 and WHITE for 1 in the panel, forcing me to export in the Text or C file to see the digits values. Why not display 0 and 1?
4. After obtained the binary string, to convert it in decimal number I have to use an external tool, like Rapid Table site - I think should be easy to include the resulting decimal number by selecting the sequence in the "chessboard" view, in a tooltip or somewhere else.
5. Seems to me there is a un unexpected behaviour, maybe "a bug"? If instead to export in a file, I select the black/white boxes, I can copy and then paste, which is great! But for some unknow reason, copy/paste miss the last bit.
[image]

**METHOD 2**
[image]

Here I have to enter manually the "Zero Point" (meaning is clear, I do it), but it also ask me for the "Clock source" / "Symbol rate" and suggest me 26 Hz..?!?
If I click "Sample" it tell me Symbol rate is too small. I have to enter a different value rather than the one suggested? (even if 26 Hz seems to me really little...) - but how to measure the expected value?

-------------------------------------------------------------------------------------------------------------------

In any case METHOD 1 works - maybe a little more tricky for that simple purpose, but works.
Evidently this tools is like a Ferrari but this is a too simple task, which would be fine for an 80s Fiat's car.

Thanks a lot for pointing me in the direction - I will continue to discover your fantastic tool.
Francesco