BatchDrake/SigDigger #262: Build from source - QT6 lrelease not found

Pny, 2024-11-23:
During build from source on Ubuntu 22.04 the system has now started looking for /usr/lib/qt6/bin/lrelease which has never been present on my computer... This might have been started after installing Qt6 packages?

How do I fix this? Something more I need to install, create a symlink, or force SigDigger to use Qt5 istead? I believe lrelease is related to Qt Linguist, which I do have installed (both Qt5 and Qt6 versions).

BatchDrake, 2024-11-23:
That's odd, I build SigDigger with Qt6 regularly. Perhaps there are some
packages missing? How did you install Qt6?

Pny, 2024-11-23:
Thanks for fast reply! :-)

At first my computer had a some Qt6 packages installed and the old Qt creator provided by Ubuntu, but after I realized this didn't compile, I first installed even more Qt6 packages using apt-get and later installed the latest Qt creator using their online install. Cleared the Sigdigger folders, redownloaded the Sigdigger source code from github once more, and still can not successfully build due to lrelease missing.... 

(I, (at least try to...), use the "develop" tag of SigDigger and the other components).

Do you know if the /usr/lib/qt6/bin/lrelease file is needed and I need to fix my Qt6 install, or is Qt6 looking for lrelease at the wrong path?

BatchDrake, 2024-11-23:
lrelease is a Qt Linguist tool needed for internationalization, perhaps
that's the package you need. Truth is, I usually install Qt6 from the
official Qt installer, and I am not sure how Ubuntu splits Qt into several
packages.

PS have in mind I merged all projects (sigutils, suscan, suwidgets and
SigDigger) to master not too long ago, and that's now the current
development branch. Make sure you have all the latest changes before
building the software.

Pny, 2024-11-23:
Oh, I didnt realize I can use the master branch, I just saw the note now after reading your comment above... (I reused my own build scripts from last time I built this some time ago, and didnt notice this had changed). 

I will give the master branch a try (on monday)... 

Thanks!

F5OEO, 2025-04-08:
Use qt6-l10n-tools package under debian based system.