BatchDrake/SigDigger #263: Ubuntu 24.4 on RP5 build issues

ranornit, 2024-12-10:
I tried all methods of building per the SigDigger_user_manual (AppImage, Building from sources, and BLSD) and failed. First, libvolk1-dev is libvolk-dev now. Second, "git clone https://github.com/BatchDrake/SuWidgets.git" and not "git clone https://github.com/BatchDrake/SuWidgets". Third - you have to create .git folder in every cmake folder of sigutils. Forth - smoothpsd.c in blsd-dir/build-root/sigutils/src/sigutils has an error in missing declaration of unsigned int i in line 167 or something like this.
And finaly - make fails, see attached logerr.
Please guide me how to install a working package on RP5 running Ubuntu 24.4

Thanks, Ran
[make-28691-stderr.log](https://github.com/user-attachments/files/18075384/make-28691-stderr.log)

ranornit, 2024-12-10:
missing declaration of unsigned int i in line 186

BatchDrake, 2024-12-10:
Hi,

Debian Sid over here, recently updated git. What git version are you using? I can clone all repos without issues:

```
% git --version
git version 2.39.5
% git clone https://github.com/BatchDrake/SuWidgets.git
Cloning into 'SuWidgets'...
remote: Enumerating objects: 1729, done.
(...)
```

Just checked your error log. Look:

```
make: /usr/lib/qt5/bin/lrelease: No such file or directory
```

`lrelease` is what it is used to build the translation files. This is a recent change to SigDigger and I assumed (wrongly, it seems) that it should be bundled to the `qt5-default` package. Try again installing `qttools5-dev-tools`.

Also, have in mind that Qt5 has reached EoL in 2023, consider building with Qt6 instead.

Regarding this error:

```
missing declaration of unsigned int i in line 186
```

I believe I have just found the cause and fixed it. Still, if this error has been triggered, it means that CMake is failing to find `volk`, which makes me think that perhaps Ubuntu has changed the name of the pkg-config module too.  Can you run this command and tell me what it prints?

```
% pkg-config --modversion volk
``

ranornit, 2024-12-10:
Thanks:
Yes, git clone https://github.com/BatchDrake/SuWidgets.git works. I just
pointed out that in the user guide it is "git clone
https://github.com/BatchDrake/SuWidgets
<https://github.com/BatchDrake/SuWidgets.git>"
As for Qt5 ot 6 - QT6 do not have locations, and I'm using it elsewhere.
I'll try again installing qttools5-dev-tools.

My git version is 2.43.0
pkg-config --modversion volk: 3.1

‫בתאריך יום ג׳, 10 בדצמ׳ 2024 ב-9:37 מאת ‪Gonzalo José Carracedo
Carballal‬‏ ***@***.***‬‏>:‬

ranornit, 2024-12-10:
installing qttools5-dev-tools solved the problems. I installed it in
somehow different way than proposed, but it worked eventually. Thanks!!

‫בתאריך יום ג׳, 10 בדצמ׳ 2024 ב-9:37 מאת ‪Gonzalo José Carracedo
Carballal‬‏ ***@***.***‬‏>:‬

ranornit, 2024-12-10:
> You are welcome!
>
> If you don't mind, I am going to keep this issue opened as a reminder for
> SigDigger 0.4 documentation. The fact that you could not build this almost
> out of the box means that the documentation is not clear enough and that
> some steps are outdated.

No problem, just let me add that qmake6 did not work for me, qmake did.

‫בתאריך יום ג׳, 10 בדצמ׳ 2024 ב-13:27 מאת ‪Gonzalo José Carracedo
Carballal‬‏ ***@***.***‬‏>:‬