BatchDrake/SigDigger #265: MacOS Suscan Build Issue

IanStewart2001, 2025-01-06:
I am trying to build SigDigger on MacOS and am following the PDF instructions. When its gets to suscan, I run "make" and get the following error. I have been trying for a while but haven't made headway. I believe it might be related to MacOS not natively supporting pthread_barrier_t for parallel programming. Any help is very much appreciated. I am MacOS Ventura 13.0

1 warning generated.
[ 24%] Building C object CMakeFiles/suscan.dir/cli/parse.c.o
[ 24%] Building C object CMakeFiles/suscan.dir/cli/datasaver.c.o
[ 25%] Building C object CMakeFiles/suscan.dir/analyzer/device/discovery.c.o
[ 25%] Building C object CMakeFiles/suscan.dir/analyzer/device/facade.c.o
[ 26%] Building C object CMakeFiles/suscan.dir/analyzer/device/properties.c.o
[ 26%] Building C object CMakeFiles/suscan.dir/analyzer/device/spec.c.o
[ 27%] Building C object CMakeFiles/suscan.dir/analyzer/device/impl/multicast.c.o
In file included from {path}suscan/analyzer/device/impl/multicast.c:32:
In file included from {path}suscan/./analyzer/analyzer.h:34:
{path}suscan/analyzer/inspsched.h:77:3: error: unknown type name 'pthread_barrier_t'
  pthread_barrier_t  barrier; /* Inspector barrier /
  ^
1 error generated.
make[2]: ** [CMakeFiles/suscan.dir/analyzer/device/impl/multicast.c.o] Error 1
make[1]:  [CMakeFiles/suscan.dir/all] Error 2
make:  [all] Error 2

BatchDrake, 2025-01-07:
Being reproduced in CI (note to myself: https://github.com/BatchDrake/suscan/actions/runs/12608654261/job/35141326376), fixing it ASAP.

BatchDrake, 2025-01-07:
Hi,

The issue was actually in Suscan, but it should be fixed now. Rebuild everything from Suscan and try agan.

Thanks,