BatchDrake/SigDigger #267: Sigdigger starts to run and then quits - DragonOS + FobosSDR

waterwin, 2025-01-28:
After setting the parameters for the Soapy connected FobosSDR in Sigdigger it shows very shortly on top that FobosSDR is running. After a few seconds Sigdigger stops like it is setting itself off.

I have setup FobosSDRas a Soapy device in latest DragonOS.

Are there any logs I can study to find causes ?

Do I need to install a new build for Sigdigger to try as well ?

Thanks
Erwin
PE3ES/F4VTQ

73

BatchDrake, 2025-01-30:
Sure, just go to the View Menu -> Log messages (Ctrl + L). No need to install any new releases.

waterwin, 2025-01-31:
Thanks, log lines found. When Sigdigger freezes after a few minutes there are no new lines in the log ...

1738322307.638257,Information,,setEnabled:194,Remote control enabled: listening on port 1234...
1738322310.537701,Information,,run:83,SigDigger 0.3.0 loaded.
1738322323.88965,Information,source,suscan_source_adjust_permissions:942,Source does not support native DC correction, falling back to software correction
1738322323.89519,Information,source,suscan_source_adjust_permissions:944,DC correction strategy: continuous