BatchDrake/SigDigger #271: Build error building SuWidgets on Raspberry Pi

InspiringCode, 2025-02-14:
Hello Gonzalo José, 

I tried to build SigDigger with blsd on a Raspberry Pi Zero 2 W with Raspberry OS Desktop but I get the following error:

```
Try: make -j 4
CWD: /home/daniel/blsd-dir/build-root/SuWidgets
WaveView.cpp: In member function ‘void WaveView::drawWave(QPainter&)’:
WaveView.cpp:463:39: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 2 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
  463 |               "Processing waveform (%ld%% complete)",
      |                                     ~~^
      |                                       |
      |                                       long int
      |                                     %lld
  464 |               100 * m_lastProgressCurr / m_lastProgressMax);
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        uint64_t {aka long long unsigned int}
GLWaterfall.cpp: In member function ‘void GLWaterfallOpenGLContext::initialize()’:
GLWaterfall.cpp:304:12: error: ‘GL_MULTISAMPLE’ was not declared in this scope; did you mean ‘GL_MULTIPLY’?
  304 |   glEnable(GL_MULTISAMPLE);
      |            ^~~~~~~~~~~~~~
      |            GL_MULTIPLY
GLWaterfall.cpp:306:12: error: ‘GL_LINE_SMOOTH’ was not declared in this scope; did you mean ‘GL_LINE_WIDTH’?
  306 |   glEnable(GL_LINE_SMOOTH);
      |            ^~~~~~~~~~~~~~
      |            GL_LINE_WIDTH
GLWaterfall.cpp:307:10: error: ‘GL_LINE_SMOOTH_HINT’ was not declared in this scope
  307 |   glHint(GL_LINE_SMOOTH_HINT, GL_NICEST);
      |          ^~~~~~~~~~~~~~~~~~~
GLWaterfall.cpp:308:12: error: ‘GL_POINT_SMOOTH’ was not declared in this scope
  308 |   glEnable(GL_POINT_SMOOTH);
      |            ^~~~~~~~~~~~~~~
GLWaterfall.cpp:309:10: error: ‘GL_POINT_SMOOTH_HINT’ was not declared in this scope
  309 |   glHint(GL_POINT_SMOOTH_HINT, GL_NICEST);
      |          ^~~~~~~~~~~~~~~~~~~~
GLWaterfall.cpp: In member function ‘void GLWaterfallOpenGLContext::render(int, int, int, int, float, float)’:
GLWaterfall.cpp:620:16: error: ‘GL_ALL_ATTRIB_BITS’ was not declared in this scope; did you mean ‘GL_ALL_BARRIER_BITS’?
  620 |   glPushAttrib(GL_ALL_ATTRIB_BITS); // IMPORTANT TO PREVENT CONFLICTS WITH QPAINTER
      |                ^~~~~~~~~~~~~~~~~~
      |                GL_ALL_BARRIER_BITS
GLWaterfall.cpp:620:3: error: ‘glPushAttrib’ was not declared in this scope
  620 |   glPushAttrib(GL_ALL_ATTRIB_BITS); // IMPORTANT TO PREVENT CONFLICTS WITH QPAINTER
      |   ^~~~~~~~~~~~
GLWaterfall.cpp:626:3: error: ‘glLoadIdentity’ was not declared in this scope
  626 |   glLoadIdentity();
      |   ^~~~~~~~~~~~~~
GLWaterfall.cpp:690:3: error: ‘glPopAttrib’ was not declared in this scope
  690 |   glPopAttrib();
      |   ^~~~~~~~~~~
make[1]: *** [Makefile.Release:799: release/GLWaterfall.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:42: release] Error 2
```

I also tried `sudo apt-get install  qttools5-dev-tools` but it didn't make any difference. I've also attached the stdout file. 

I would be really grateful if you could give me a tip or two how to proceed. Would be really eager to try out your software. Thanks!

Best regards,
Daniel

[blsd-stdout.txt](https://github.com/user-attachments/files/18803526/blsd-stdout.txt)

BatchDrake, 2025-02-14:
Hi! What Qt version do you have installed in your pi?

InspiringCode, 2025-02-14:
```
daniel@zero2w:~ $ qmake --version
QMake version 3.1
Using Qt version 5.15.8 in /usr/lib/arm-linux-gnueabihf
```

```
daniel@zero2w:~ $ qtchooser -list-versions
4
5
default
qt4-arm-linux-gnueabihf
qt4
qt5-arm-linux-gnueabihf
qt5
daniel@zero2w:~ $ qtchooser -print-env
QT_SELECT="default"
QTTOOLDIR="/usr/lib/qt5/bin"
QTLIBDIR="/usr/lib/arm-linux-gnueabihf"
```

```
daniel@zero2w:~ $ cat /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5 | strings | grep "Qt 5"
Qt 5.15.8 (arm-little_endian-ilp32-eabi-hardfloat shared (dynamic) release build; by GCC 12.2.0)
```

I hope this helps!

InspiringCode, 2025-02-16:
I asked ChatGPT about this error and its answer was basically:

> The problem is that you are compiling code that relies on deprecated desktop OpenGL features that are not present in OpenGL ES. The Raspberry Pi Zero 2 W, by default, uses an OpenGL ES implementation rather than full desktop OpenGL. OpenGL ES is a streamlined version that omits many of the legacy functions you’re trying to use. You’ll need to adjust your code or build environment accordingly.

Does this make sense? Didn't people already compile SigDigger for Raspberry Pi several times?

BatchDrake, 2025-02-16:
I wouldn't even bother asking ChatGPT about it, particularly when dealing with stuff that is outside general knowledge. Chances are that it will make up a response and add more confusion to the issue.

Some time ago we did a refactor of the OpenGL Waterfall and we know it works with Qt6. This error looks like being related to a missing header that is implicitly included in Qt6 but perhaps not in Qt 5. I will look into it.

BatchDrake, 2025-02-16:
Well, it turns out that ChatGPT was right in this case. I am going to add some `#define`s to disable those calls when compiling against OpenGL ES.

BatchDrake, 2025-02-16:
I have just updated the code, do you mind pulling from `master` and compiling it again?

InspiringCode, 2025-02-16:
Thank you very much for looking into it, I really appreciate it! SuWidgets successfully builds now. The script is building SigDigger now - takes quite a while on a Zero 2W, but I will keep you updated.

BatchDrake, 2025-02-16:
Glad to hear that! Still, have in mind that if SigDigger runs under this setup, it will run using the QPainter Waterfall by default. I don't know if the OpenGL Waterfall is going to work, but if you still want to test it, go to settings (Ctrl + S), GUI tweaks tab, check the OpenGL waterfall, Accept changes and restart SigDigger normally (by closing the window, not hitting Ctrl+C!).

InspiringCode, 2025-02-17:
I managed to compile SigDigger now, but it still doesn't startup:

```
daniel@zero2w:~/blsd-dir/SigDigger $ ./SigDigger 
/home/daniel/blsd-dir/SigDigger/bin/SigDigger: locale en_GB not found, using English as default
qt.svg: Cannot open file ':/icons/nodemod.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/nodemod.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/am.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/am.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/fm.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/fm.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/lsb.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/lsb.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/usb.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/usb.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/raw.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/raw.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/ask-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/ask-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/fsk-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/fsk-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/psk-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/psk-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/rms-inspector.svg', because: No such file or directory
qt.svg: Cannot open file ':/icons/rms-inspector.svg', because: No such file or directory
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 191, resource id: 0, major code: 140 (Unknown), minor code: 20
```

But this might well be a problem with my Raspberry Pi setup. I am using NoMachine to connect to the Pi and disabled Wayland. But it might also well be that the Raspberry Pi Zero 2 W is just too weak/slow for SigDigger.