BatchDrake/SigDigger #272: Windows Build Using a Qt Subdirs Project

simurg1728, 2025-02-16:
Hello,

I would like to contribute to the Windows builds. I have created a Qt subdirs project at https://github.com/simurg1728/SigDigger-MAIN, which includes all the required dependencies. I have also made minor modifications to the SigDigger libraries.

I am using Radioconda for third-party libraries (I know there are more elegant solutions, but this seemed like the easiest approach). I am able to build the project on the master branch and run the application on Windows. However, I am unable to detect my SDR devices with SoapySDR, so I will be looking into this issue.

I would like to know if this approach would be helpful.

BatchDrake, 2025-02-16:
Hello! I fixed Windows builds literally days ago, hope our work didn't overlap too much. I still have to make a development release for all 3 architectures, and macOS is the one being the blocker for now.

I don't know if you were aware of this but, there is a Windows build script in the Scripts folder (`Scripts/dist-mingw32.sh`) that works on GitHub CI's machines and requires MSYS2, a bunch of pacman packages and renaming the `lrelease` executable (see the full action file here: https://github.com/BatchDrake/SigDigger/blob/master/.github/actions/win32deps/action.yml). Would it be too much of a hassle if you try to repeat these steps in your machine? I can send you a prebuilt binary SigDigger for Windows if you want too.

simurg1728, 2025-02-16:
Hello,
I wasnt aware of the Windows scripts. I should have check project files. But it is good to hear Windows builds are fixed. I can check them but it may take a while.

BatchDrake, 2025-02-16:
The script is fairly automatic, you only need to clone SigDigger, install
those dependencies (once) and just call build-mingw32. It will start
cloning all the remaining dependencies (including sigutils and Suscan) and
build everything step by step. If the script fails it will output a .txt
file with the error messages you can attach to this issue.

TobiasPSP, 2025-03-24:
I just came across your project - awesome! That said, I am a Windows person, and while I am also developer, I am not familiar with github actions and the specifics about your build scripts. 

So at this point I am sadly forced to move on. If you do find the time to add Windows binaries, I believe this increases relevance and reach of your awesome work tremendously. 

At this point, I simply don't know how to compile the sources with your scripts, and like so many others, I'm looking to invest time on your software features, not on figuring out github :-)

Many thanks anyway for your awesome work.

BatchDrake, 2025-03-24:
Hi,

I've just added an automatic Windows development release: https://github.com/BatchDrake/SigDigger/releases/tag/latest

I was reluctant to make this release because only RTL-SDR and HackRF are supported, remote analyzers don't work and I can't test this as thoroughly as I'd like. However, you are not the only one reminding me that I should release something. So here it is.

0.4.0 is on its way, I'm trying my best but my availability is particularly reduced this year (PhD and stuff). Hope this changes at some point next year.

simurg1728, 2025-03-25:
Thats excellent. Thanks a lot. I would like to help with Windows scripts, but unfortunately I have very little time. I think this issue can be closed now.

closed 2025-03-25 by simurg1728