BatchDrake/SigDigger #274: New release??

melroy89, 2025-03-14:
Current release is from 2022 (v0.3.), we are now in 2025. What about creating a new release 🚀 ?

TobiasPSP, 2025-03-24:
I just came across [this](https://x.com/BatchDrake/status/1489195043525177344) in an effort to maybe find a Windows release, however the 2022 post seems to have linked to a category instead a build, so the link seems gone.

I know free work is free, and there's nothing for us to ask for, but after having watched your awesome presentation from FOSDEM 2021, I'd love to see you release a Windows build, even if not all features are fully functional at this point. 

To ship is to choose, but please ship ;-)

TobiasPSP, 2025-03-25:
Greatly appreciated, many thanks! I'll keep you updated with my findings, may take a day or two. I start by using RTL-SDR. Eventually, I have also access to a RX-888 which I could try, however it's notorious for its driver issues.

jlpoolen, 2025-05-06:
For what it is worth: https://github.com/BatchDrake/sigutils/issues/115 "Time For A New Release above 0.3.0?"  I was unaware of this Issue when I logged the aforesaid issue.